Sensirion/arduino-sht #9: Error compiling in SparkFun ESP32

JuliGonzalez, 2020-01-11:
I have downloaded the library via .zip and imported it into Arduino IDE as I did with a few and those ones worked correctly.

When I import this library into the sketch and i try to run the code I always get the same error and I'm not able to fix it.

Here is the error i'm getting:
```
Arduino:1.8.10 (Mac OS X), Tarjeta:"ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x1440824]

goroutine 1 [running]:
github.com/arduino/arduino-cli/legacy/builder.ResolveLibrary(0xc000118800, 0xc00020816b, 0xb, 0xc00020816b)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/resolve_library.go:64 +0x1d4
github.com/arduino/arduino-cli/legacy/builder.findIncludesUntilDone(0xc000118800, 0xc0004dac60, 0x1578ca0, 0xc000180cd0, 0xc00048eb70, 0xc0002a5500, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/container_find_includes.go:358 +0x3db
github.com/arduino/arduino-cli/legacy/builder.(*ContainerFindIncludes).Run(0x1bacff0, 0xc000118800, 0x1bacff0, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/container_find_includes.go:152 +0x4e6
github.com/arduino/arduino-cli/legacy/builder.runCommands(0xc000118800, 0xc0006cbc20, 0x20, 0x20, 0x1, 0x5d6a0, 0x1bae720)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:210 +0xdf
github.com/arduino/arduino-cli/legacy/builder.(*Builder).Run(0xc0000a3e70, 0xc000118800, 0x1bacef0, 0xc0000937e0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:117 +0xd28
github.com/arduino/arduino-cli/legacy/builder.RunBuilder(...)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:226
main.main()
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-builder/main.go:398 +0x800
/private/var/folders/hb/gnqzxwy578d8j85vcl588lym0000gn/T/AppTranslocation/45A947C1-A4C3-4560-A826-6C5A53E6653F/d/Arduino.app/Contents/Java/arduino-builder retornó 2
Error compilando para la tarjeta ESP32 Dev Module.
```

Sorry for showing the error in Spanish but is how I firstly configured the IDE.

I tried reinstalling the library and other stuff but couldn't made it work.

winkj, 2020-02-06:
Is this inside the IDE, or a build with jenkins? I see a lot of paths that include `/home/jenkins/`, so wondering where this is coming from

winkj, 2020-02-06:
Also, I find similar reports for 1.8.10, with a recommendation to roll back to 1.8.9: https://forum.arduino.cc/index.php?topic=638271.0
https://mansfield-devine.com/speculatrix/2019/09/arduino-running-hard-to-stand-still/

Also, 1.8.11 is out, might be worth checking this out

closed 2020-02-06 by winkj