Sensirion/arduino-sht #10: I2C communication without clock stretching

pifatech, 2020-02-02:
Hi,

I have used this library with a Sensirion sensor without success. There is a initialization issue with the sensor.

Since the clock stretching has been disabled a delay has been added between the sensor measurement request and the measurement readout. 
The current implemantation ignores this delay and send a I2C restart between the measurement request and the readout.

I propose to replace this:

https://github.com/Sensirion/arduino-sht/blob/ed9485cf2e062e692cd3d7ddf31dd02e9ab9230c/SHTSensor.cpp#L70

by:

`if (Wire.endTransmission() != 0) { `

Thank you and best regards.

winkj, 2020-02-03:
Hi there,

Removing the "false" makes sense for the reasons you describe. 

Just to be sure, does this change fix the problem you've mentioned in the beginning of the report?

Thanks, Johannes

pifatech, 2020-02-03:
Thank you for your reply.
Yes, after removing the "false" I can initialise the sensor and read the values.

TechnicalLee, 2021-02-03:
I have this same issue causing many random `Error in readSample()` messages with the example code. The fix in the first message worked for me, no more errors. I suggest the developer makes that revision. Thanks @Lubnitex.

closed 2021-06-10 by winkj