Sensirion/arduino-sht #11: All examples generate a compile error

bfcooper, 2020-02-06:
I'm running Arduino on Windows 10 and I get a compile error on all three of the example programs.
Here is the error message:
**Arduino: 1.8.10 (Windows 10), Board: "Arduino/Genuino Uno"

panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0x0 pc=0x7855fb]

goroutine 1 [running]:
github.com/arduino/arduino-cli/legacy/builder.ResolveLibrary(0x118de5a0, 0x118c8bbd, 0xb, 0x118c8bbd)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/resolve_library.go:64 +0x14b
github.com/arduino/arduino-cli/legacy/builder.findIncludesUntilDone(0x118de5a0, 0x118f5d80, 0x88ed00, 0x11888090, 0x11a1d328, 0x11804d00, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/container_find_includes.go:358 +0x3b6
github.com/arduino/arduino-cli/legacy/builder.(*ContainerFindIncludes).Run(0xdebc68, 0x118de5a0, 0xdebc68, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/container_find_includes.go:152 +0x3e5
github.com/arduino/arduino-cli/legacy/builder.runCommands(0x118de5a0, 0x11873e1c, 0x20, 0x20, 0x409a01, 0xffffff, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:210 +0xbd
github.com/arduino/arduino-cli/legacy/builder.(*Builder).Run(0x11873f40, 0x118de5a0, 0xdebbb4, 0x11806bb0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:117 +0xb9c
github.com/arduino/arduino-cli/legacy/builder.RunBuilder(...)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:226
main.main()
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-builder/main.go:398 +0x6d2
C:\Program Files (x86)\Arduino\arduino-builder returned 2
Error compiling for board Arduino/Genuino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
**

winkj, 2020-02-06:
all the error output here is from builder.go, so it seems like it's a problem with the toolchain rather than with the example.

Can you reproduce this outside of jenkins, in the Arduino IDE?

winkj, 2020-02-06:
Also, I find similar reports for 1.8.10, with a recommendation to roll back to 1.8.9: https://forum.arduino.cc/index.php?topic=638271.0
https://mansfield-devine.com/speculatrix/2019/09/arduino-running-hard-to-stand-still/

Also, 1.8.11 is out, might be worth checking this out

closed 2020-02-06 by winkj