Sensirion/arduino-sht #13: Unstable readings on the SHT85 sensor

wavewolf, 2020-06-23:
I can't get stable readings on the SHT85 sensor. 
I use the example code "sht-autodetect" and this is what is printed by the serial monitor:

[image]

And sometimes he doesn't even give me readings, he just gives me the error.
What could be going on?

winkj, 2020-06-23:
Which Arduino board are you using for this test? And are there any other sensors connected to the I2C bus?

wavewolf, 2020-06-23:
I forgot to say, I'm sorry, I'm using the ESP32 development board.
It may not be entirely compatible?

winkj, 2020-06-23:
I've previously tested on ESP32 - an Adafruit HUZZAH32 - and didn't experience issues, but will do more testing.

Since you had issues on and off, one possibility could be a connection issue. Is this using a third party sensor board, or is it a direct connection with soldered wires?

wavewolf, 2020-06-23:
I have sensirion SHT85 module wired to the ESP32 module.
 If you say that it works correctly in your tests, I can think that it is a problem with my cables or that the sensor is defective.

wavewolf, 2020-06-24:
I have news, I have continued to do tests and it works fine if I restart the module a couple of times. The first two times, it usually gives me that error, but the third time, it manages to start and print the values. I'm still investigating.

erickkill, 2020-11-23:
Hi,

How wiring the sensor on ESP32 pins?

What the sequence?

TechnicalLee, 2021-02-03:
Try what the author of #10 suggested as a fix, that solved all the sensor errors for me as well.

winkj, 2021-03-08:
Closing for now; if this is a reproducible issue, please feel free to reopen

closed 2021-03-08 by winkj