Sensirion/arduino-sht #16: [remove Compiler Warning]: caused by unused Variable in SHTSensor.h

hasenradball, 2021-06-05:
Hi,

I have  a proposal to remove a silly Compiler warning in line 160 of SHTSensor.h:

> see here:
``` 
          virtual bool setAccuracy(SHTSensor::SHTAccuracy newAccuracy) {
                 return false;
          }
```

> causes a silly compiler warning which can be prevented like:
```
          virtual bool setAccuracy(SHTSensor::SHTAccuracy newAccuracy) {
                (void) newAccuracy;
                return false;
          }
```

winkj, 2021-06-07:
Looks good, thanks for reporting this. While harmless, libraries should not generate warnings. Alternatively, we could used ```__attribute__((unused))```

hasenradball, 2021-06-07:
Hi thanks for watching 😀.
How will it look like with the __attribute__((unused))?

winkj, 2021-06-07:
It would look like this:
```
virtual bool setAccuracy(__attribute__((unused)) SHTSensor::SHTAccuracy newAccuracy) {
    return false;
}
```

I've briefly tried to recreate the compiler warning in Arduino IDE, but failed to do so. Could you share the setup you used to get this warning so we can verify that it's fixed? Thanks

winkj, 2021-06-07:
Alternatively, we can simply omit the parameter name:
```
virtual bool setAccuracy(SHTSensor::SHTAccuracy /* newAccuracy* /) {
    return false;
}
```
This should fix it too

hasenradball, 2021-06-08:
Hi Johannes,

I use Platform IO where I am able to set the Warning level in GCC compiler like that:
```
; Debug Flag Serial or Serial1
build_flags =
    -Wall
    -Wextra
```

The sesult is like that:
```
In file included from D:\PIO_libs\cSensorSHT31/cSensor_SHT31.h:7,
                 from D:\PIO_libs\cSensorSHT31\cSensor_SHT31.cpp:8:
.pio\libdeps\esp12e\arduino-sht/SHTSensor.h: In member function 'virtual bool SHTSensorDriver::setAccuracy(SHTSensor::SHTAccuracy)':
.pio\libdeps\esp12e\arduino-sht/SHTSensor.h:160:51: warning: unused parameter 'newAccuracy' [-Wunused-parameter]
  160 |   virtual bool setAccuracy(SHTSensor::SHTAccuracy newAccuracy) {
```

hasenradball, 2021-06-08:
I tried to set the warning level in Arduino IDE, but I didn't found where.

hasenradball, 2021-06-08:
```
virtual bool setAccuracy(__attribute__((unused)) SHTSensor::SHTAccuracy newAccuracy) {
    return false;
  }
```
works also fine
:-)

hasenradball, 2021-06-08:
Hi Johannes,
kann ich noch eine Frage stellen?
Ich habe einen ESP8266 mit einem SHT31 Sensor.

Ich nutze folgendes Setup:

```
namespace SHT {
  class Sensor_SHT31 {
    private:
      uint8_t _sda_pin, _scl_pin;
      SHTSensor sht = SHTSensor(SHTSensor::SHT3X);

    public:
      bool sensor_available;
      bool send_data2DB;
      struct Messung {
        time_t zeit;
        float Messwert1;
        float Messwert2;
        float Messwert3;
        float Messwert4;
      } Data;
      explicit Sensor_SHT31(uint8_t sda_pin, uint8_t scl_pin);
      bool setup(void);
      void measure();
  };
}
```
```
bool SHT::Sensor_SHT31::setup() {
  // SDA, SCL
  Wire.begin(_sda_pin, _scl_pin);
  delay(100);
  sensor_available = sht.init();
  DBG__PRINT(">>> SHT31D-Sensor - init(): ", sensor_available, "\n\n");
  // only supported by SHT3x
  sht.setAccuracy(SHTSensor::SHT_ACCURACY_MEDIUM);
  return sensor_available;
```
Das `sht.init()` liefert mir einen `true` Wert obwohl kein Sensor angeschlossen ist.

Ist das normal?

winkj, 2021-06-08:
@hasenradball das ist konsistent mit dem Code wenn ein Sensor-Typ angegeben wird, aber nicht unbedingt sinnvoll. Werde das auch gleich anschauen

winkj, 2021-06-08:
@hasenradball both issues are addressed in pull request #17

closed 2021-06-09 by hasenradball