Sensirion/arduino-sht #24: connect sensor to 2nd I2C bus

borlowsky, 2022-04-20:
Hi, thanks a lot for this excellent library. I have a project where I want to connect two SHT30-sensors to an ESP32 and wonder whether it's possible to connect an SHT30 to the 2nd I2C bus of the ESP32. Did not find out whether it's possible to pass the bus as an argument to init() or to the constructur... Thanks for any hint!

winkj, 2022-04-20:
Hi there, 

This is indeed possible, the init() function allows to use an optional TwoWire parameter: https://github.com/Sensirion/arduino-sht/blob/master/SHTSensor.cpp#L285.

At this point, the autodetect does not work on the second bus, but I just merged #23 which will address this. To make it work with the current version, you will have to create the SHTSensor with a sensor type, like so:
```
SHTSensor sht1(SHTSensor::SHT3X);
SHTSensor sht2(SHTSensor::SHT3X);

// ...
 Wire.begin();
// ...

sht1.init(<insert bus 1 here>);
sht2.init(<insert bus 2 here>);
```

You can start with the multiple-sht-sensors.ino example as a starting point, change both sensors to be SHT3X (assuming they are), and then add the two busses as arguments to ```init(...)```

I will close this, feel free to reopen if you run into any issues!

Best, Johannes

borlowsky, 2022-04-21:
Thanks a lot for your quick reply! Don’t know why I did not get it to work yesterday, but now it’s fine!
Cheers, Boris.

winkj, 2022-04-21:
Thanks a lot for testing!

I just pushed a new release (1.2.2) which includes the fix for autodetect, so changes are the following will work too:

```c++
SHTSensor sht1, sht2; // autodetect

// ...
 Wire.begin();
// ...

sht1.init(/* insert bus 1 here */;
sht2.init(/* insert bus 2 here */;