Sensirion/arduino-sht #26: Bootloop

myctindustry, 2022-09-21:
Hi,

I have followed the code as posted by [borlowsky](https://github.com/borlowsky)

Using a nodemcu-32s and uploaded the sketch below.

After upload I am receiving a bootloop.

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0

Not sure where I went wrong on the code.

#include <Wire.h>
#include "SHTSensor.h"

#define I2C1_SDA 16                                  // I2C1 data pins
#define I2C1_SCL 17                                  // I2C1 data pins
#define I2C2_SDA 18                                 // I2C2 data pins
#define I2C2_SCL 19                                  // I2C2 data pins

SHTSensor sht1(SHTSensor::SHT3X);
SHTSensor sht2(SHTSensor::SHT3X);

TwoWire I2C1 = TwoWire(0);
TwoWire I2C2 = TwoWire(1);

void setup() {
  // put your setup code here, to run once:
  
  I2C1.begin(I2C1_SDA, I2C1_SCL, 10000);
  I2C2.begin(I2C2_SDA, I2C2_SCL, 10000);
  
  Wire.begin();
  Serial.begin(9600);
  delay(1000); // let serial console settle

  // init on a specific sensor type (i.e. without auto detecting),
  // does not check if the sensor is responding and will thus always succeed.

  // initialize sensor on 1st bus
  sht1.init(I2C1);

  // initialize sensor on 2nd bus
  sht2.init(I2C2);
}

void loop() {
  // put your main code here, to run repeatedly:
  // read from first sensor
  if (sht1.readSample()) {
    Serial.print("SHT1 :\n");
    Serial.print("  RH: ");
    Serial.print(sht1.getHumidity(), 2);
    Serial.print("\n");
    Serial.print("  T:  ");
    Serial.print(sht1.getTemperature(), 2);
    Serial.print("\n");
  } else {
    Serial.print("Sensor 1: Error in readSample()\n");
  }

  // read from second sensor
  if (sht2.readSample()) {
    Serial.print("SHT2:\n");
    Serial.print("  RH: ");
    Serial.print(sht2.getHumidity(), 2);
    Serial.print("\n");
    Serial.print("  T:  ");
    Serial.print(sht2.getTemperature(), 2);
    Serial.print("\n");
  } else {
    Serial.print("Sensor 2: Error in readSample()\n");
  }
 

  delay(1000);
}

winkj, 2022-09-21:
Does the same error occur with the official samples, e.g. `sht-autodetect`? I'd start there as a baseline, and then enable the second sensor support line by line.

One thing that I would recommend is checking the return values for the calls to `init()` to make sure the sensors were properly detected, although the library should be robust enough to handle this without causing bootloops.

myctindustry, 2022-09-22:
Hi winkj

sht-autodetect does not cause bootloop but will return "error in readsample" regardless of 1 or 2 sensors.

sht-multiple-sht-sensor will return "error in readsample 2" if only 1 sensor is plugged in. And error in readsample when both sensors are plugged in.

The multiple-sht-sensor has a comment regarding address pins. However, the SHT-85 only has 4 pins, no separate address pin.
// Sensor 1 with address pin pulled to GND
// Sensor 2 with address pin pulled to Vdd

I will try with the auto-detect line by line but is it even worth moving forward if no sensors are being auto-detected?

Thanks

winkj, 2022-09-22:
> I will try with the auto-detect line by line but is it even worth moving forward if no sensors are being auto-detected?

Just to be sure, did you adjust the Wire configuration for the `sht-autodetect` sample?

Since all the readSamples seem to fail here, I personally would try first with a simple I2C scanner, like shown here: https://playground.arduino.cc/Main/I2cScanner/. Of course, you may have to adjust the pins to the bus you're using. I'd focus on one I2C bus first, and make sure that the sensor is being discovered properly. This should rule out any wiring issues. If this is working, troublesh

The address of the SHT85 is going to be fixed at the default (0x44), so you shouldn't have to do do any additional configuration for that.

myctindustry, 2022-09-22:
Hi winkj

Sorry, Just tested again with sht-autodetect sample to verify.
With just a single sensor it is in fact showing sensor readings. 
If multiple sensors are plugged in to the same i2c bus then "error in readsample" occurs.

I will modify the wire configuration and see if it provides a reading on 2 separate i2c buses as well as the i2c scanner if it fails.
Thanks

winkj, 2022-09-22:
Glad to hear, and keep us posted once you get it to work.

It may make sense to try the second bus on it's own first, too - i.e. make sure that both sensors work individually, then move on to running them at the same time.

myctindustry, 2022-09-24:
Hi winkj

Got it working with the following code:
SHT-85 x 2
NodeMCU-32s

Thanks for your support.

#include <Wire.h>
#include <SHTSensor.h>
#define I2C2_SDA 4
#define I2C2_SCL 5

SHTSensor sht1;
SHTSensor sht2;

void setup() {
  // put your setup code here, to run once:
  
  Serial.begin(9600);
  Wire.begin();
  Wire1.begin(I2C2_SDA, I2C2_SCL);
  Wire.setClock(100000);
 
  delay(1000); // let serial console settle

  if (sht1.init()) {
      Serial.print("init(): success\n");
  } else {
      Serial.print("init(): failed\n");
  }

  if (sht2.init(Wire1)) {
      Serial.print("init(): success\n");
  } else {
      Serial.print("init(): failed\n");
  }

}

void loop() {
  // put your main code here, to run repeatedly:

  if (sht1.readSample()) {
      Serial.print("Bottom:\n");
      Serial.print("  RH: ");
      Serial.print(sht1.getHumidity(), 1);
      Serial.print("\n");
      Serial.print("  T:  ");
      Serial.print(sht1.getTemperature(), 1);
      Serial.print("\n");
  } else {
      Serial.print("Error in readSample1()\n");
  }

    if (sht2.readSample()) {
      Serial.print("Top:\n");
      Serial.print("  RH: ");
      Serial.print(sht2.getHumidity(), 1);
      Serial.print("\n");
      Serial.print("  T:  ");
      Serial.print(sht2.getTemperature(), 1);
      Serial.print("\n");
  } else {
      Serial.print("Error in readSample2()\n");
  }
  delay(5000);
}

winkj, 2022-09-26:
Awesome, thanks for reporting back. We have a todo item to add a sample for multiple sensors to this library.

closed 2022-09-26 by winkj