Sensirion/arduino-sht #27: First Reading High with autodetect example

floatAsNeeded, 2022-10-07:
Hi!

I just wanted to note an issue when using the example sht-autodetect with the SHT40. I'm not sure what the cause of the issue is, but when I use the SHT40 the first reading is not a proper value and gradually cools down while reading.  It looks like the heater goes on for a short period or something like that. This issue doesn't happen when using the SHT3x series. 
This is easily fixed by just using this line instead:

SHTSensor sht(SHTSensor::SHT4X);

Here is an example of what the issue looks like in the Serial:

```
15:46:14.342 -> init(): success
15:46:14.389 -> SHT:
15:46:14.389 ->   RH: 70.45
15:46:14.389 ->   T:  29.50
15:46:15.360 -> SHT:
15:46:15.360 ->   RH: 67.22
15:46:15.405 ->   T:  22.48
15:46:16.376 -> SHT:
15:46:16.376 ->   RH: 66.68
15:46:16.421 ->   T:  22.06
15:46:17.392 -> SHT:
15:46:17.392 ->   RH: 67.05
15:46:17.438 ->   T:  21.95
15:46:18.410 -> SHT:
15:46:18.410 ->   RH: 67.57
15:46:18.410 ->   T:  21.93
15:46:19.431 -> SHT:
15:46:19.431 ->   RH: 67.94
15:46:19.431 ->   T:  21.87
15:46:20.479 -> SHT:
15:46:20.479 ->   RH: 68.36
15:46:20.479 ->   T:  21.85
15:46:21.487 -> SHT:
15:46:21.487 ->   RH: 68.65
15:46:21.487 ->   T:  21.83
15:46:22.472 -> SHT:
15:46:22.472 ->   RH: 68.90
15:46:22.519 ->   T:  21.81
15:46:23.501 -> SHT:
15:46:23.501 ->   RH: 69.06
15:46:23.501 ->   T:  21.77
```

Any Ideas on how to possibly fix that?

winkj, 2022-10-07:
Thanks for the report. I was using an ESP32 with an Adafruit SHT40 board, and could reproduce this successfully. An interesting case for sure!

> Any Ideas on how to possibly fix that?
Your workaround - to initialize SHTSensor as an SHT4X, rather than using the automatic detection - seems to work fine. 

Other than that, my guess is that since the SHT3x is on the same I2C address, attempting to communicate with a SHT3x causes the SHT4x to do something unexpected. This would also explain why the user who contributed the SHT4x support had to add an extra delay here to make it work (since the SHT4x was in an strange state): https://github.com/Sensirion/arduino-sht/blob/master/SHTSensor.cpp#L317

To verify this, I switched out the order for the autodetect sensors here (https://github.com/Sensirion/arduino-sht/blob/master/SHTSensor.cpp#L277) to try to detect the SHT4x first, and have since not been able to recreate the issue. Unfortunately, I didn't have an SHT3x available to make sure it didn't break that. 

If you have a moment, I'd appreciate if you could try that as well and let me know if this fixes it for your setup, too.

winkj, 2022-10-07:
Here's the array I'm using now:

`const SHTSensor::SHTSensorType SHTSensor::AUTO_DETECT_SENSORS[] = {
  SHT4X,
  SHT3X,
  SHT3X_ALT,
  SHTC1
};
const f`

floatAsNeeded, 2022-10-07:
Thank you very much for the detailed reply! I will try that array with both the SHT3x and 4x and report back.

floatAsNeeded, 2022-10-07:
I just tried with my setup and now it's working perfectly fine for both the SHT4x series and the SHT3x series using autodetect.

I think the issue as you noted was regarding the SHT4x that it's a bit "slow" to set up properly, so this issue is solved by putting it first on the array and it doesn't affect the stability at the start of the SHT3x

Thank you very much for your help! I guess we can close the issue and modify the library to change the order in the array.

floatAsNeeded, 2022-10-09:
Regarding my last question, I found a way to get the SHT class of the sensor by calling **sht.mSensorType** if someone wants to know.
It will return a number and that number will have to be looked up to know which sensor you're talking to.

`  enum SHTSensorType {
    /** Automatically detect the sensor type (only i2c sensors listed above) */
    AUTO_DETECT,
    // i2c Sensors:
    /** SHT3x-DIS with ADDR (sensor pin 2) connected to VSS (default) */
    SHT3X,
    SHT85,
    /** SHT3x-DIS with ADDR (sensor pin 2) connected to VDD */
    SHT3X_ALT,
    SHTC1,
    SHTC3,
    SHTW1,
    SHTW2,
    SHT4X
  };`

For example, the SHT4x will return 8 and SHT3x 1

winkj, 2022-10-10:
I'll reopen this while we're testing the fix

floatAsNeeded, 2022-10-22:
Just wanted to leave a comment that I'm testing the setup since then and no issue encountered with both the SHT3x and SHT4x

I'm testing them also together via connecting a multiplexer and using the same address I2C for both

PR #32 (2023-11-14): change order of auto detection
Since the SHT3x and SHT4x share addresses, there's an issue when an SHT4x is connected to the bus and we're probing for an SHT3x: The result is that the first reading of the SHT4x is off.

The other direction works (probing for a 4x when a 3x is connected), as such we're simply changing the order of detection

Fixes #27
closed 2023-11-14 by winkj