Sensirion/arduino-sht #35: SHT40 Raspberry Pi Pico Arduino Earle Phillhower

TheLennyLeonard, 2024-03-18:
Hello, I have a problem with your library in arduino with Earle Phillpower library. I use your SHTSensor library, i think correctly but still not getting responses from SHT40. Can you please help me? (it is correctly wired - power 3,3V, pull up for SDA,SCL - 2,2kOhm resistors)

the SHT40 is connected to
GP14 --> PIN 19 on the Pico
GP15 --> PIN 20 on the Pico
`
#include <Wire.h>
#include "SHTSensor.h"

SHTSensor sht(SHTSensor::SHT4X);

float actualTemperature = 99.9;
float actualHumidity = 99.9;

void setup() {
pinMode(14,INPUT);
pinMode(15,INPUT);
Wire1.setSDA(14);
Wire1.setSCL(15);

Wire1.begin(0x44);

Serial.begin(9600);
while (!Serial)
delay(10);

if (sht.init()) {
Serial.print("init(): success\n");
} else {
Serial.print("init(): failed\n");
}
sht.setAccuracy(SHTSensor::SHT_ACCURACY_MEDIUM);
}

void loop() {
if (sht.readSample()) {
actualHumidity = sht.getHumidity();
Serial.print(actualHumidity, 2);
Serial.print("\n");
Serial.print(" T: ");
actualTemperature = sht.getTemperature();
Serial.print(actualTemperature, 2);
Serial.print("\n");
Serial.print("Reading SHT40 Data...OK\n");
} else {
Serial.print("Error in readSample()\n"); // this is being displayed in serial monitor - so sht.readSample() doesnt read anything
}
delay(1000);
}
`

winkj, 2024-03-18:
on a cursory check, I believe you didn't initialize the SHT library on Wire1. Could you change the call of `sht.init()` to `sht.init(Wire1)` and try again?

winkj, 2024-03-18:
could you print the complete output? `sht.init()` should fail if the sensor is not detected ("init(): failed"), before readSample(). It should have either printed an error before (defaulting to `Wire`) or after the change to use `Wire1`, unless the Pi pico platform is working differently than other Arduinos.

As a further test, running I2CScanner helps: https://playground.arduino.cc/Main/I2cScanner/. This should detect a sensor responding on the SHT40's I2C address `0x44`. This should help narrow down the root cause

TheLennyLeonard, 2024-03-18:
yeah, I was getting init(): failed responses...so I tried the I2C scanner and even the scanner cant find it...
I am afraid that because it is a pico not arduino...it will never gonna work.¨

here is the code:
`#include <Wire.h>

const byte PICO_I2C_ADDRESS = 0x44;
const byte PICO_HUMIDITY_PRECISION = 0xFD; // high precission
const byte PICO_I2C_SDA = 12;
const byte PICO_I2C_SCL = 13;

 
void setup()
{
  pinMode(14,INPUT);
  pinMode(15,INPUT);
  Wire.setSDA(PICO_I2C_SDA);
  Wire.setSCL(PICO_I2C_SCL);
  Wire.begin();
 
  Serial.begin(9600);
  while (!Serial);             // Leonardo: wait for serial monitor
  Serial.println("\nI2C Scanner");
}
 
 
void loop()
{
  byte error, address;
  int nDevices;
 
  Serial.println("Scanning...");
 
  nDevices = 0;
  for(address = 1; address < 127; address++ )
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
 
    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");
 
      nDevices++;
    }
    else if (error==4)
    {
      Serial.print("Unknown error at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");
 
  delay(5000);           // wait 5 seconds for next scan
}
`
getting No I2C devices found...